UPetersen/LibreMonitor #2: Sketchs from Arduino

hferreira, 2016-11-23:
Hello,

I built a limiter myself.

Some questions:
1. The sketch will work on this setup ?
2. What are the differences between Simblee and RFDuino ?
3. What model do you used in your project ?

Regards,
Hugo.

UPetersen, 2016-11-23:
Hi Hugo,

My first LibreMonitor was built with an RFDuino and worked well. I then switched to a Simblee because it is smaller and also allows for FOTA (which I have not really used yet). So much for the differences (for more on this see the respective websites)

If you build a LibreMonitor with an RFDuino instead of a Simblee, you have to change some code in the LibreMonitor.ino file and in libUBP.cpp in LibreMonitorArduinoLibrary.zip. Basically you have to substitute „Simblee“ by „RFDuino“. I just looked quickly through the code. It should be the following two changes:

1. Use the appropriate includes for RFDuino instead of those fore Simblee
2. Change the Simblee commands to the respective RFDuino commands, e.g. change SimbleeBLE.advertisementData = "data" to RFDuinoBLE.advertisementData = "data"; etc.

Please refer to the manuals of [Simblee](https://www.simblee.com/Simblee%20Functions%20Reference%20V1.0.pdf) and [RFDuino](http://www.rfduino.com/wp-content/uploads/2014/03/rfduino.ble_.programming.reference.pdf) for details and possible differences of these commands. 

If you do these code changes using #ifdefs in the code and submit a pull request I could check them in and it would be of help for others who want to build a LibreMonitor with an RFDuino, too. One could switch quite easily between Simblee or RFDuino by changing some #define commands and the includes in the info files.

Regards
Uwe

hferreira, 2016-11-23:
Ho, I thought that Simblee was a superset of RFDuino and larger than the original one, using the same chip on his core but I check and see that RFDuino is 15mmx15mm and Simblee is 7mmx10mm.

hferreira, 2016-11-23:
Hi Uwe,

Ho, I thought that Simblee was a superset of RFDuino and because of that was larger than the original one, using the same chip on his core but I checked and I saw that RFDuino is 15mmx15mm and Simblee is 7mmx10mm.
It's a superset yes but smaller, right ?
So, on this case it's more wise to use Simblee instead of the original RFDuino.

From the RFDuino website (so, the same should be applied for Simblee), the Arduino sketch should work of of the box for RFDuino, but you have some dependencies from Simblee, why ?
The original Limitter.ino can't work ?

Regards,
Hugo.

UPetersen, 2016-11-24:
Hi Hugo,

I am not sure, what you are aiming at and what you mean with "original Limitter.ino". Is it the Limitter.ino of the [LimiTTer Project](https://github.com/JoernL/LimiTTer) by JoernL? If so, the answer is no. The Limitter.ino will not work with LibreMonitor because LimiTTer and LibreMonitor transfer different sets of data between the reader device and the respective smart phone application. 

If you want to use your device with LibreMonitor follow my suggestions above and you can display the data with the LibreMonitor iOS application. 

If you want to build a LimiTTer and display the data with the corresponding application you have to make adaptions to its LimiTTer.ino to make it work with an RFDuino, but I cannot tell you what exactly has to be changed. You could try and ask this question on the respective [Gitter blog](https://gitter.im/JoernL/LimiTTer) and maybe get some help. 

Regards, 
Uwe

hferreira, 2016-11-24:
Hi Uwe,

I already have my own fork of the Limitter.ino working with my own Android App.
I thought that my ino sketch could work out of the box with RFDuino, because RFDuino is a kind of clone of Arduino + Bluetooth module but for what you are saying it's not just that easy, so I will have  review the code or start over again from your ino sketch.

The ideia behind using RFDuino (or Simblee) is to reduce the size and connections of the hardware.
By luck, seems that most of RFDuino/Simblee pins can be soldier next to BM019.

Regards,
Hugo.

UPetersen, 2016-11-25:
Hi Hugo,

wish you good luck. 

Regards
Uwe

closed 2016-11-25 by UPetersen