UPetersen/LibreMonitor #3: Prepare the RFDuino/Simblee board

hferreira, 2016-12-14:
Hello,

After several weeks of delaying, I finally got my RFDuino  (similar to yours).
How did you safely remove the plastic from the pins to attach on the NFC board ?

Regards,
Hugo.

hferreira, 2016-12-14:
Nervermind. It was easy.

UPetersen, 2016-12-14:
Hi Hugo,
I have been working on a RFduino version, too. I just committed my current code for your ease. It is still work in progress and not yet tested, but probably a better starting point than the original code in this repository. (I seem to have a loose contact somewhere. So I could not receive valid data from the BM019, yet, but data transfer to the iOS app works.). You'll find some instructions in the comment section of the ino-File.
Regards 
Uwe

hferreira, 2016-12-14:
Thank you very much.
It's a good starting point.

hferreira, 2017-01-09:
Hello,

I was out for a while and now I came back and return to this project.
Starting with your INO file and after compile and upload I got on my Arduino IDE debug window informing that the setup was successful (also my Android App can be notified about the connection) and it fails when to retried data from BM019 but this is probably the issue that you mention.

Regards,
Hugo.

closed 2018-04-25 by UPetersen

basma37, 2019-05-03:
I have a compile error when trying to upload a code on rfduino

basma37, 2019-05-03:
Error is cannot run program(runtime.tools.arm-none-eabi-gcc.path)...

UPetersen, 2019-05-04:
This does not look like a LibreMonitor specific error to me, but rather like something general regarding rfduino. Have you managed to run any code on your rfduino so far?