UPetersen/LibreMonitor #4: Light always fixed

hferreira, 2016-12-18:
Hello,

I already connected all wires on a breadboard (RFDuino + NFC-BM019).
I saw a lot of compile issues on your RFDuino ino version so I downloaded the original one and replaced all Simblee references to RFDuino. Compiled and uploaded to the hardware.

My Android App (that previous worked with Limitter, so the main difference should be the UUID address) is never notified but I also noticed that the NFC chip have the light always fixed.
On my previous version (based on Limitter), the light is off when is sleep mode and turn on when is reading.

PS: I already checked the wire connections over and over again.

Regards,
Hugo.

UPetersen, 2016-12-18:
Hi Hugo,

### Wiring

  With the LibreMonitor wiring the red light on the BM019 will shine permanently. That is the intended setup using the on-board voltage regulator of the BM019. It is directly connected to the lipo and the 3.3 V output of the regulator serves as input to the RFduino. 

### Ino-File and LimiTTer application

  Using the LibreMonitor.ino file you will NOT be able to communicate with an app that expects data from a LimiTTer, as already noted in [issue2](https://github.com/UPetersen/LibreMonitor/issues/2). LibreMonitor and LimiTTer use similar hardware but are totally different on the software side since they use

- a different protocol to transfer data and 
- a different set of data that is transferred between device and smartphone application. 

Thus, the LibreMonitor.ino only works with the LibreMonitor iOS application. If you want to use a different application you have to either change the ino file accordingly to provide LimiTTer-like data or adjust your application to work with the LibreMonitor data set.

I am still not sure, If I correctly understand what your intention is. Do you want to make your Android application behave as the LibreMonitor iOS application and use the same protocol and data set? Or do you want to leave your app as it is and make the RFduino setup behave as a LimiTTer?

Regards
Uwe

hferreira, 2016-12-19:
Thank you for the explanation about the wiring.
OK, I see. It's always shinning (in opposite to Limitter) consuming a little more energy but not that much because only periodically BM019 is activated to read, so it's in a hibernate mode, correct ?

"Using the LibreMonitor.ino file you will NOT be able to communicate with an app that expects data from a LimiTTer"
Yes, I know. I have my own App and not xDrip. The base BT connection should be the same. Only the UUUID can be different and the chosen protocol that can be easily changed on INO or on my App.

I can now receive notifications on my App from the device trying to connect but still not fire the characteristics. I will investigate more.

Thank you very much for your time.

Regards,
Hugo.

UPetersen, 2016-12-19:
Correct, BM019 is periodically set to hibernate mode. After that the Simblee/RFduino is set to ultra low power mode (Simblee_ULPDelay(SLEEP_DURATION) /  RFduino_ULPDelay(SLEEP_DURATION) with SLEEP_DURATION set to two minutes). After the two minutes the Simblee/RFduino wakes up, wakes up the BM019, both do their job and then go to sleep again... 

Note, using the Simlcee/RFDuino_ULPDelay(...)-command the Simblee/RFduino remains connected to the Smartphone application, but energy consumption is low. I like this mode since it can be conveniently used as a permanent indication for a working connection.

##### Data Transfer

I use [this ](https://github.com/cconway/RFduinoUBP) code to have a simple to use protocol. You can give it a c-struct with content longer than 20 bytes. The data is encapsulated using [SLIP](https://de.wikipedia.org/wiki/Serial_Line_Internet_Protocol) and a crc8 is appended to ensure data integrity. The packet is then transferred to the iOS application. There the crc8 is checked and if it is okay the packet is decoded back to the original struct. I made some changes to the [original code](https://github.com/cconway/RFduinoUBP) to make it work with the 344 bytes, so you may want to use the code from the LibreMonitor repository. 

The c-structs are defined in [data_types.h](https://github.com/UPetersen/LibreMonitor/blob/master/LibreMonitor/Bluetooth/data_types.h). The main Freestyle Libre data that is transferred are the 344 bytes in "allBytes" of the struct "AllBytesDataType" that correspond to Blocks 0x00 to 0x2A when you read a sensor with [tagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=de) or some other tag reader. You can follow the code in [SensorData.swift](https://github.com/UPetersen/LibreMonitor/blob/master/LibreMonitor/Model/SensorData.swift) and corresponding files on how to get desired information such as history and trend raw glucose data.

##### Bluetooth

The main part of the bluetooth code on the application side is in [SimbleeManager.swift](https://github.com/UPetersen/LibreMonitor/blob/master/LibreMonitor/Bluetooth/SimbleeManager.swift). In there you find how I handle the UUIDs, services and characteristics. When a data packet is received it is then handled in [this](https://github.com/UPetersen/LibreMonitor/blob/master/LibreMonitor/ViewControllers/BloodSugarTableViewController.swift#L356) part of BloodSugarTableViewController.swift. This is where the c-structs are received as "payloads" and the data is extracted.

Hope this provides some help for your own application. It took me quite some time to understand the bluetooth stuff and build this application. Wish you good luck.

Of course you can use any other more simple protocol for data transfer.

Regards, 
Uwe