UPetersen/LibreMonitor #6: Switch lipo charger example

clvsjr9, 2016-12-21:
Could you indicate a switch lipo charger URL to appreciate?

UPetersen, 2016-12-21:
I only have experience with the first of the two Adafruit lipo chargers that are mentioned in the readme and the "SparkFun LiPo Charger Basic", which worked fine, too. I have no experience with any other lipo and thus cannot recommend any other.

clvsjr9, 2016-12-21:
The readme file presents the follow text: 

Switch (optional but helpfull if you mount a lipo charger).

Could you point to me any product to this item? (I'd like to use one)

UPetersen, 2016-12-22:
A switch is helpful to disconnect the BM019 and Simblee/RFDuino from power, if the device needs a reset. LibxMike used a setup with a switch, as you can see in the picture in the readme. I haven't used a switch by myself, yet, and that is why I did not provide a link and cannot really recommend one from own experience. I always disconnect and reconnect the JST plug of the lipo if a reset is needed. I made a quick search and think the following [switch](http://www.exp-tech.de/mini-slide-switch-3-pin-spdt-0-3a-3-pack) should be okay and serve your needs.

clvsjr9, 2016-12-24:
Very thanks!

closed 2016-12-24 by clvsjr9