UPetersen/LibreMonitor #7: Thank you

hferreira, 2017-01-10:
I created this issue only to thank you UPeterson.

My current INO file it's very different but finally working now !
Your layout helped me go to there.
I had to remap the original pins: http://www.rfdigital.com/wp-content/uploads/2014/03/rfduino.ble_.programming.reference.pdf and probably it's one difference comparing to Simblee.
I wanted the pins on the same configuration as yours because fits perfectly on BM019.

Best regards,
Hugo.

UPetersen, 2017-01-10:
Hi Hugo,

great to hear, that you got your device working. Congrats.

Regarding the remapping I actually put some information in the [LibreMonitor.ino](https://github.com/UPetersen/LibreMonitor/blob/master/LibreMonitor.ino#L26) as guidance on how to modify the variant.h file and a note on this in the [README.md](https://github.com/UPetersen/LibreMonitor/blob/master/README.md). But thanks to your remarks I will make this more prominent in the future. 

Just out of curiosity: What kind of smart phone app do you use? Is it based on LibreMonitor or did you develop it yourself or?

Best regards,
Uwe

hferreira, 2017-01-10:
Hi Uwe,

Thank you.

Yes, the remapping was missing for me and it's a key feature for this project because allows to connect to NFC directly and reduce wires.

My own App: https://play.google.com/store/apps/details?id=pt.solidsoft.appforlibre
This App allows to read directly from NFC (like others) and I changed enough to also read from BLE reusing the same Widget and chart.

Best regards,
Hugo.