UPetersen/LibreMonitor #8: Battery test

hferreira, 2017-01-16:
Hello,

I made 2 tests with a full charged batteries:
- LiPo 3.7v 240mAh - After 24h with 3 m sync period consumed only 35%.
- LiPo 3.7v 120mAh - After 18h with 5 m sync period consumed 90% and the system sttoped to communicate.

Do you think this is a normal behaviour ?

Regards,
Hugo.-

UPetersen, 2017-01-16:
Hi Hugo,

I have been using a 150 mAh lipo for a long time. I put the libreMonitor on ever morning and take it off when I go to bed (because I don't want to lie on it and possibly brake it, since I still don't use a case) and then I charge it over night. With a reading every two minutes even on long days with up to 18 or 20 hours it still worked when I took it of. And this even though the Simblee keeps a permanent bluetooth connection to the iPhone between the readings (with "Simblee_ULPDelay" command). 

I put your and my values in the following table. Unfortunately I don't have a means to measure the remaining capacity for my lipo so I just assumed 100 percent. 

|        | Hugo 1 (RFduino)| Hugo 2 (RFduino) | Uwe (Simblee) |
| :--- | :---:  |     :---:      | :---: |
|lipo energy [mAh] | 240 | 120 | 150 |
|used energy [%] | 35 | 90 | 100 |
|used energy [mAh] | 84 | 108 | 150 |
|usage time [h] | 24 | 18 | 18 |
|delay between readings (min) | 3 | 5 | 2 |
|reading frequency [1/h] | 20 | 12 | 30 |
|number of readings in usage time | 480 | 216 | 540 |
|energy consumption per reading [mAh] | 0,175 | 0,5 | 0,28 |

So your large lipo performs very well in comparison to mine but your smaller lipo could do better. Overall it looks reasonable to me. Since I don't know what changes you possibly made: Please note that this comparison is only valid on the assumption that you read and transfer the same amount of data. 

Regards,

Uwe

hferreira, 2017-01-16:
Hi Uwe,

Thank you very much for your comparison table.
I will also test for 2 minutes.
Do you find that there is any side effect with 1 minute besides the battery faster consume.

Yes I used always the same sample data to avoid mistaken results.
I agree that my large lipo performs very well (more that what was expected), however my smaller lipo is not enough (I wanted 24h / one day charge) for the same 3 minutes (at most).
Shouldn't the smaller lipo performance as 50% comparing to the bigger one ?
I mean, for he same 24h (3 minutes delay), consume 70% energy ?

Regards,
Hugo.

UPetersen, 2017-01-17:
Hi Hugo,

I never tried a one minute interval. I just don't think it is necessary. I like the two minutes pretty much. Three minutes seems reasonable, too, but five minutes is just to long for me. 

I am not sure if the one minute interval will have side effects. 

Regarding side effects: Rarely, but it sometimes happens that I don't get a response from LibreMonitor any more and the Freestyle Libre Reader also refuses to provide values. Usually, after a while everything works fine again. And usually I take the LibreMonitor of when this happens. I do not know why this happens and I don't know if this is caused by LibreMonitor at all. But for sure I know that I had observed this with sensors before having started to use LibreMonitor, too. Same with sensors that completely failed. Just got two such cases recently with sensors that were very close to their expiry date. But also did have this in the early days without LibreMonitor. 

To sum up: If you are keen on a one minute interval you have to try yourself. I personally would not expect problems, but who knows. And yes, I'd be interested in the results. 

Lipo: Yes, I would have expected the small lipo to perform more equally, too. Maybe it has a different discharge capability (if this is of influence) and it could also be just a bad lipo. I guess that to be really sure about lipo performance and requirements more systematic tests will be necessary.

Regard
Uwe

hferreira, 2017-01-18:
Hi Uwe,

"I never tried a one minute interval. I just don't think it is necessary. I like the two minutes pretty much. Three minutes seems reasonable, too, but five minutes is just to long for me."
I agree with you.

"I am not sure if the one minute interval will have side effects."
Stop working for example but I will use 2 or 3 minutes.

"Regarding side effects: Rarely, but it sometimes happens ... "
I'm doing tests and experience exactly the same behaviour (I see that I'm not alone here :))

"Lipo: Yes, I would have expected the small lipo to perform more equally, too. Maybe it has a different discharge capability (if this is of influence) and it could also be just a bad lipo. I guess that to be really sure about lipo performance and requirements more systematic tests will be necessary."
I have been reading about discharge capability and in theory yes, it can influence a little but I don't believe that much. I bet in got bad 120mAh Lipo and a very good 240mAh one.
I'm expecting to receive tomorrow a new 150mAh (with the same discharge capability as the 240mAh and let's see how performe this one).

Regards,
Hugo.

hferreira, 2017-01-19:
A new test with a new LiPo 150mAh.
After 24h with 3m interval it keep working with full charge (of course it's not possible and a read battery level issue) but working.

A will run a new 24h test with the same 150mAh LiPo but now with 2m interval.

kozyretti, 2017-12-28:
What about safety of LiPo using? 
The device is fixed on the arm all the time. Is this not dangerous?
I mean a sudden ignition.

UPetersen, 2017-12-28:
I do use a regulated Lipo and I haven't had any bad experience so far. Neither have I heard of any bad experience from similar projects like [LimiTTer](https://github.com/JoernL/LimiTTer).