UPetersen/LibreMonitor #9: A series of build errors

erikstr, 2017-03-09:
I am trying to build the iOS app straight out of the fresh downloaded sources, but getting a lot of errors. Here is a screenshot of them. What am I doing wrong?
[image]

UPetersen, 2017-03-09:
Hi Erik,

have you installed cocoapods? There are some third party libraries used, that require cocoapods. For installation [https://cocoapods.org](https://cocoapods.org).

erikstr, 2017-03-09:
No, I hadn't :( sorry that I didn't pay attention to this. 
However, even with the Charts lib installed, I still get compile errors :((
[image]
[image]

erikstr, 2017-03-09:
Here are the full details:
[image]

UPetersen, 2017-03-09:
Mmmh, that leaves me scratching my head. I just downloaded the code from GitHub myself, ran "pod install" in the directory, opened the workspace file and compiled with simulator vor iPhone SE and it runs. I use Xcode 8.2.1.
[image]

UPetersen, 2017-03-09:
Have you tried to clean the build (Product and then "alt" - "Clean Build Folder". Sometimes that helps.

UPetersen, 2017-03-09:
Also worth a try (last answer): [swiftc failed with exit code 1](http://stackoverflow.com/questions/41203153/swiftc-failed-with-exit-code-1).

UPetersen, 2017-03-09:
Or this one: [ERROR swiftc failed with exit code 1](http://stackoverflow.com/questions/38496582/error-swiftc-failed-with-exit-code-1).

erikstr, 2017-03-09:
8.2.1 here as well. Yes, I clean every time. I suspect, that I am doing something wrong with the Charts lib. The instruction there says "Drag the Charts.xcodeproj to your project". However you can't drag just this file leaving the whole folder structure behind, so I download the zip, unpack it and include the lib from there to the target.

erikstr, 2017-03-09:
I'll switch to my MacBook now and start over from scratch ...

UPetersen, 2017-03-09:
I never dragged the charts lib. I installed it via cocoapods. Try this: Once you have cocoapods installed, just download the complete LibreMonitor-project again. Open a shell, change into the LibreMonitor directory and there type "pod install". Then open the workspace file with Xcode and compile.

erikstr, 2017-03-09:
Success! 
It turns out that the downloaded charts lib I previously linked is for macOS only, no iOS. Now the app compiles and runs on the iPhone. The RFduino connects also fine but I am getting no readings :( What worries me even more is that no BM019 ID is shown as when there is no connection between the RFduino and the BM019 module. The ino sketch compiled and loaded fine. How could I investigate this problem now? I checked the wiring, looks good. But the BM019 LED is lighting red. Is this normal?

erikstr, 2017-03-09:
Here is a screenshot:
[image]

UPetersen, 2017-03-10:
This looks good. The right light on the BM019 is right. And you have a bluetooth connection to the RFduino. Now you should check, wether you get valid signals on the Arduino side, best by using the serial monitor. Set the 

`#define DEBUG `

and you should see some hex values. To reduce your waiting times, you can reduce the sleep time for tests, e.g. by setting it to 20 s:

`const uint64_t SLEEP_DURATION = 20000;  // Duration of Simblee ultra low power mode in ms`

Another important thing is to have the SPI wiring correct. If you wired as depicted in the pictures, you have to change the RFduino pins in software. This is done in the variant.h file. In my case this file is located (dependent on the Simblee/RFduino version) in

/Users/[my user name]/Library/Arduino15/packages/Simblee/hardware/Simblee/1.0.0/variants/Simblee or
/Users/[my user name]/Library/Arduino15/packages/RFduino/hardware/RFduino/2.3.3/variants

In this file set the defines for the SPI pins as follows:

```
///         #define PIN_SPI_SS           (3u)
///         #define PIN_SPI_MOSI         (5u)
///         #define PIN_SPI_MISO         (4u)
///         #define PIN_SPI_SCK          (6u)
```

I also sent you a private note on gitter.