UPetersen/LibreMonitor #10: Translation for raspberry

F5OEO, 2017-09-27:
Thx for your amazing work. I am working on a raspberry pi version, your code is my basis to communicate with FreeStyle. I wonder if you have links for more details about the protocol. Glimp (closed source) can indicate for example if sensor is outdated...

My goal is to send datas directly to nightscout (not yet enough info to do that).

UPetersen, 2017-09-27:
The information if a sensor is expired resides in byte #4 of the header, see [SensorData.swift](https://github.com/UPetersen/LibreMonitor/blob/swift3.2_xcode9/LibreMonitor/Model/SensorData.swift#L47). 

All what I know about how the Freestyle Libre data can be interpreted is actually represented in the [SensorData class](https://github.com/UPetersen/LibreMonitor/blob/swift3.2_xcode9/LibreMonitor/Model/SensorData.swift#L1) and the other classes in the [model directory](https://github.com/UPetersen/LibreMonitor/tree/swift3.2_xcode9/LibreMonitor/Model). I do not know about any complete and good documentation, other than Pierre Vandevenne's [Blog](http://type1tennis.blogspot.de). (Start Reading somewhere in late 2014.)

Regarding nightscout: I implemented a nightscout upload in the swift3.2_xcode9 branch, using code from [NightscoutUploadKit](https://github.com/ps2/rileylink_ios/tree/master/NightscoutUploadKit). The relevant code portions are located in the [NightscoutUploader](https://github.com/UPetersen/LibreMonitor/tree/swift3.2_xcode9/LibreMonitor/NightscoutUploader) directory. I call it [around here](https://github.com/UPetersen/LibreMonitor/blob/swift3.2_xcode9/LibreMonitor/ViewControllers/BloodSugarTableViewController.swift#L525) to upload all history data. It works quite well. This branch also has a [settings view](https://github.com/UPetersen/LibreMonitor/blob/swift3.2_xcode9/LibreMonitor/ViewControllers/SettingsViewController.swift) to enter nightscout site and API secret.

F5OEO, 2017-09-27:
Greatly thx to all these links...you save hours of googling and github search ! As I am an old style guy, I have surely to translate to C but ressources is there ! Again a great thx to your contribution. Hope to publish something working (but ugly as my usual style). Let me know if I can send you detail question out of an "issue github scope" directly by mail. evaristec @ gmail dot com