UPetersen/LibreMonitor #11: Can't find "Charts" module

JJBarata, 2018-04-25:
Hi! I downloaded the latest files and followed all their guidelines, however, when trying to build the application, I'm getting some warnings and a strange error that refers to importing the module "Charts", which it does not find. Here are the screenshots. What am I doing wrong? Tks.

[image]

[image]

[image]

JJBarata, 2018-04-26:
Thanks for reply!

I used master branch on swift 4 and xcode 9.3 environment. I did all steps that you suggested (installed cocoapods, ran "pod install" on project folder, opened "LibreMonitor.xcworkspace", cleaned build folder, added "Charts.framework" on Embedded Binaries and tested on iPhone SE) but when I test, appear all issues. I did it right?

I using Simblee hardware with BM019 module.

I will test with 25 Nov. 2017 branch and I will post here the results.

Thank you very much and congratulations for your awesome project!!

JJBarata, 2018-04-26:
Well, I ran now and return one error, at least! I noticed that all errors occur because of the Charts. I made some changes on Pods and Charts build settings like prints bellow.

[image]

[image]

[image]

Thanks!!

JJBarata, 2018-04-26:
I made it work. I ran on my iPhone 7 Plus. I installed Charts with Carthage, as you indicated. Now I will configure the Simblee and the BM019 to be able to scan the FL.

[image]

Thank you very much!!

UPetersen, 2018-04-26:
😄

closed 2018-04-26 by UPetersen