UPetersen/LibreMonitor #12: App no show glucose level and others informations

JJBarata, 2018-05-07:
Hi Uwe,

As I said before, I was able to run the app from the version you indicated, but I can not get the app to show the blood glucose level, blocks, status and start of the sensor, and the graph with the measurements. I see in the Arduino IDE that the BM019 reads the sensor and sends the packets and information so much that I can see the ID of the BM019, the sensor ID and the ambient temperature and voltage, as in the image below. I'm not using lipo, I'm using the USB cable on the Mac.

[image]

I did the tests with some sensors, already expired and with an active sensor that my son is using.

If you need more information, just talk.

I'm grateful for your help.

Thank you,

Jorge

UPetersen, 2018-05-08:
Hi Jorge,

so the data transfer between simblee and iOS works, and the BM019 is connected to the simblee, since its ID is transferred, too. This is already quite good. Now the question is why you do not receive any data from the sensor. Having the simblee attached to USB while running is okay, I did so often, too. As a next step I suggest you try to print out some data with the serial monitor of the Arduino IDE (if this what you use as IDE) and submit it here. You should uncomment the 

`
 #define DEBUG
`

to receive the information from all the print commands in the code. Could you also give me a link to the arduino code you are using?

UPetersen, 2018-05-09:
Mmh. Data in serial monitor looks good. FRAM data is read properly. So the question is, why this data does not arrive in the app. Next step would be to set a stop point in the debugger in [here](https://github.com/UPetersen/LibreMonitor/blob/485aed2ed195b788dd1d83d54b68caf70610adc5/LibreMonitor/ViewControllers/BloodSugarTableViewController.swift#L406) and see what cases are run through. This is the code section where the different data types are received. According to the screenshot you provided above, 

`case .SYSTEM_INFORMATION_DATA:`

works and 

`case .BATTERY_DATA:`

works, although the battery value should be higher, and

`case .IDN_DATA:` 

also works. The question is, why 

`case .ALL_BYTES: `

seems not to work. Please run the debugger and check weather the `case .ALL_BYTES: ` code section is called or not. (If not, we have to double check the data types: files data_types.h and constants.h must be the same in the arduino code and the app code.) To get the os_log outputs in the debugger, you have to comment out the lines after [here](https://github.com/UPetersen/LibreMonitor/blob/485aed2ed195b788dd1d83d54b68caf70610adc5/LibreMonitor/AppDelegate.swift#L63). We'll then continue.

JJBarata, 2018-05-10:
I agree!

The code in the constants.h and data_types.h files are the same in Arduino and in the application.

I made debbug and the output log and prints are in the folder I shared earlier.

I noticed that many of the data could not be read, as I show in the prints.

If you need more information, I am at your disposal.

Thank you!

UPetersen, 2018-05-10:
The prints look good to me. It is the case of system information data received and all is going well. The question is, why the case with the fram data (allBytes) is not called. Could you set a break point behind  

`case .ALL_BYTES:`

to see if it gets ever called? If this is not the case I recommend we continue this by phone. Sent you a PM on gitter.