UPetersen/LibreMonitor #19: Comments regarding Libre OOP Investigation

JtxGit, 2019-04-30:
Trying to understand how things work has always fascinated me and I found your "Libre OOP Investigation" very interesting.

You said:

>

>  To sum up: We have
> 
>     a linear glucose response by the LibreOOP algorithm with respect to raw glucose variations, when raw temperature is kept constant, and
>     a linear glucose response by the LibreOOP algorithm with respect to raw temperature variations, when raw glucose is kept constant.

In this case we can use a bi-linear function (as is, for example, used in linear finite element formulations):
z = a1 + a2 x + a3 y + a4 xy

These 4 constants can (relatively easily) be found if we have 4 known values of z for given x and y.
This might give us a smaller target to aim for in trying to find some clues in the header / footer.

From your investigation, it is clear that each sensor will have different values for these a1 to a4.
Perhaps we can find promising candidates in the header or footer.

Just for the record, the "traditional" "slope" and "offset" can be calculated for each raw temperature as follows: 
offset[T] = a1 + a3(T) 
slope[T] = a2 + a4(T)

Regards
jtx

UPetersen, 2019-05-01:
Hi Jtx,

you exactly got the point. Further down in the wiki I elaborate also on this, except that I do not use a1 to a4 but rather these terms:

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; **glucose = slope(raw_temperature) * raw_glucose + offset(raw_temperature)**

where slope(raw_temperature) and offset(raw_temperature) are derived from the two equations 

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; **slope(raw_temperature) = slope<sub>slope</sub> * raw_temperature + offset<sub>slope</sub>**

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; **offset(raw_temperature) = slope<sub>offset</sub> * raw_temperature + offset<sub>offset</sub>**.

The four parameters slope<sub>slope</sub>, offset<sub>slope</sub>, slope<sub>offset</sub> and offset<sub>offset</sub> are sensor specific and thus have to be derived for each sensor at the beginning of its lifetime. I use the [SwiftLibreOOPWeb](https://github.com/dabear/SwiftLibreOOPWeb) web interface for deriving these parameters. 

All this is implemented in the swift4 branch of LibreMonitor and works pretty well. 

Regards
Uwe

JtxGit, 2019-05-08:
To experiment some more, I created a simple Android app that allows me to enter (or read from a file) the header and footer. It then can calculate 4 points (T 5816 and 7124, G 1000 and 3000 as your example), using the OOPAlorithm apk. From these it calculates a0 to a4. Now to try and recognise a pattern...
Unfortunately my available data is only from 2 sensors, so not really enough. Now, I'm wondering if there perhaps is a list of known headers and footers somewhere that could be used to create a bigger data pool to use for pattern recognition.

UPetersen, 2019-05-09:
There is no common list of headers and footers to my knowledge. But I have plenty from my own sensors, that I can provide you with. If you give me your mail address or a private message on gitter, I can sent you some data.

JtxGit, 2019-05-09:
I'm not sure how spam-sensitive leaving an email address is, so the best is to send it to android7@eko3-tek.se

(Remove all the numbers to get the real address)

Thanks in advance.

UPetersen, 2019-05-09:
Got the email and thus deleted it from your post.

dabear, 2019-05-10:
https://libreoopweb.azurewebsites.net/Home/CalibrateSensor will give you the same parameters as output. Example implementation of those parameters into a function to calculate glucose:
https://github.com/dabear/pyderivedalgo/blob/master/pyderivedalgo.py

I see you have created your own app implementing a bi-linear function. Please note that you need static input for that to work. Readouts from the libre sensor will contain historic values and trend values. Abbott uses some form of extrapolation on top of this to yield a "currentbg". My calibratesensor endpoint removes extrapolation from their algorithm.

I have both swift and c# implementation of this available.

JtxGit, 2019-05-10:
Thanks for the information, I appreciate it and will have a closer look. 

Before I got the LibreOopAlgorithm running on Android, I actually tried to use your CalibrateSensor page, but was unsuccessful, since I could not figure out what was required for "Accesstoken"... 
Anyway, now that I have the LibreOopAlorithm running, I can easily "calibrate" a set of header / and footer by calculating 4 points (A, B, C, D) for 2 RawG and 2 RawT. Then I calculate the bi-linear coefficients a1 to a4 as follows:

`
    private static double G1 = 1000;
    private static double G2 = 3000;

    private static double T1 = 5816;
    private static double T2 = 7124;

    private static double dT = (T2-T1);
    private static double dG = (G2-G1);
    private static double a4 = (A+C-B-D)/(dG*dT);
    private static double a3 = ((D-A)/dT-G1*a4);
    private static double a2 = ((B-A)/dG-T1*a4);
    private static double a1 = A-G1*a2-T1*a3-G1*T1*a4;
`

With the bi-linear function, I can then calculate the historic and trend values from the raw values.
They correspond quite well with what the LibreOopAlgorithm gives for the same input.

The fun challenge is to see if one could find some correlation between header/footer data and these "calibrated constants" for each sensor.

closed 2019-09-29 by UPetersen

soad89, 2020-08-30:
Hi Guys. 

I would like to play with OOP header/footer dependency. Can someone guide me how to get running my own OOP with a possiblity to send there data over TCP/HTTP? Is physical device still the only solution?

In addition I would like to ask something which is still unclear for me even after studing gitter related threads:
1) Is Libre 2 over NFC encrypted or not and can be explained like L1 data.
2) Does anyone tried to massively forward artifial data to OOP and have some outputs of it?
3) Is not yet clear which bytes of header/footer are the ones that changes the output?

Thanks

UPetersen, 2020-08-31:
HI soad89, 
regarding running oop own your own: please ask in the [gitter channel on oop](https://gitter.im/LibreOOPAlgorithm/Lobby) or ask @dabear for access to his web interface. You could also try [this](https://glucose.space), but I am not sure if this is the right link and how it is to be used. 
To your other questions:
1. -> Libre2 over NFC is encrypted. 
2. -> Not to my knowledge.
3. -> Yes, at least not to my knowledge. I have fiddled around a bit myself varying header and footer bytes and observing if they change the glucose results. And it seemed to me that many of those bytes do have an influence on glucose results. But I could not determine any rule or algorithm behind this.

soad89, 2020-09-01:
Hi UPetersen, 

What is a then current status of Libre2 handling? Has anyone broken the encryption so one can use Libre1 formula? Or is oop only way how to get glucose? I supose that you can pass encrypted data to oop for L2, right?

Thank you very much for your answers.

JtxGit, 2020-09-01:
I also played with the headers and footers. Unfortunately no "A-ha" moments...
Then I got oopalgorithm running on my phone, which moved playing with the raw data down on my to-do list. 

As for L2 decryption, the LibreTools (by ivalkou) repository (for IOS) looks very interesting. I'm running android, so I have not looked into it in detail.