UPetersen/LibreMonitor #23: Trend value index seems to be not in Range

deepakthakur143, 2021-01-08:
In SensorData.swift file

`let headerRange =   0..<24   //  24 bytes, i.e.  3 blocks a 8 bytes
  let bodyRange   =  24..<320  // 296 bytes, i.e. 37 blocks a 8 bytes
  let footerRange = 320..<344  //  24 bytes, i.e.  3 blocks a 8 bytes
  
  self.nextTrendBlock = Int(body[2])

  ...........

  for blockIndex in 0...15 {.                            //The trendMeasurement function logic is
            var index = 4 + (nextTrendBlock - 1 - blockIndex) * 6 // runs backwards
            let range = index..<index+6
            let measurementBytes = Array(body[range])

            .......
`

Following is the byte array read through NFC from the Libre Sensor
`-86 45 0 0 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 **74 70 71 89 48 56 50 45 84 48 56 56 52 63 -36 73 102 -125 16 0 105 10 -64 78 20 3 -106 -128 90 0 -19 -90 26 127 90 -22 4 -27 -118 104 -46 2 -56 27 0 0 0 0 -70 40 -85 16 11 0 28 1 -35 67 -22 -60 -103 2 -44 67 -22 -40 90 2 -60 67 -22 -20 90 2 -59 67 -22 -64 -102 2 -66 67 -22 -24 27 2 -70 67 -22 -28 90 2 -75 67 -22 20 91 2 -75 67 -22 32 91 2 -82 67 -22 56 91 2 -80 67 -22 24 91 2 -87 67 -22 -8 -102 2 -26 67 -22 -52 27 2 -52 67 .......and more bytes ** `
Note: Bold ones belong to the body. 
  
`self.nextTrendBlock = Int(body[2])`, since body[2] = 71, therefore nextTrendBlock = 71.
 var index = 4 + (nextTrendBlock - 1 - blockIndex) * 6 => index = 4 + (71 - 1 - 0) * 6 = 424

Here, the index comes to be 424, but the bodySize is expected to be 296 bytes, i.e., out of index.

So I am concerned here if
a) The byte layout has changed since this code was first written.
b) Was this code developed for a fixed byte layout format.

PS: I used NFC to pull out data from the LibrePro sensor.

UPetersen, 2021-01-08:
The code ist tested for Libre1. LibrePro has a different Layout. Please search the web for information about LibrePro-Layout. Doing a quick search I found [this](https://github.com/JoernL/LimiTTer/issues/29). Thus, Libre Pro seems to have more history records than the Libre 1.

Regarding the numbers: you have negative integer values, which should not be the case. The range of the numbers should be between 0 and 255. Thus, please convert the bytes to Uint8.

deepakthakur143, 2021-01-08:
Thank you so much @UPetersen . This link would sure be helpful for me. I have been working on this for a month now, but without much success. I will explore further.

UPetersen, 2021-01-08:
No worries. Meanwhile there are quite a few other projects, that are also more up to date than LibreMonitor, some of which might also address the Libre Pro. A good starting point could be to get through the list of my starred projects and have a look at those that are a) on the libre and b) where recently updated.

closed 2021-01-09 by UPetersen